NVlabs/GraspGen #2: Asking for parallel grippers grasping pose for custom objects

khanh0812, 2025-07-19:
Many thanks for your excellent work. Could you please provide the plan for releasing parallel grippers grasping pose for custom objects, it would be helpful for my research.

adithyamurali, 2025-07-20:
If you are using a single cup or multi-suction cup gripper, the grasp data generation for custom objects is already released: https://github.com/NVlabs/GraspGen/blob/main/tutorials/TUTORIAL.md#step-1-generate-dataset

khanh0812, 2025-07-24:
@adithyamurali 
Thank you very much for your response. I am looking forward to the next release.

khanh0812, 2025-09-15:
@adithyamurali  do you have any update for releasing parallel grippers grasping pose for custom objects? Many thanks

closed 2025-10-03 by adithyamurali