NVlabs/GraspGen #8: how did you obtain the high quality point cloud?

Guptajakala, 2025-07-25:
Hi, thanks for sharing your work! The point cloud in the demo dataset looks really impressive! I'm wondering what camera did you use to obtain those high quality point cloud?

adithyamurali, 2025-07-25:
Hi @Guptajakala 

I was using the amazing work of [FoundationStereo](https://nvlabs.github.io/FoundationStereo/) to get high quality depth/point clouds. Section 4.5 of the [GraspGen paper](https://arxiv.org/pdf/2507.13097) has more details of this. Do note that in terms of hardware, we were still using a single D435 camera, though the default depth images coming from realsense were significantly worse and super noisy.

On top of FoundationStereo, I still did some basic [point cloud outlier rejection](https://github.com/NVlabs/GraspGen/blob/main/grasp_gen/utils/point_cloud_utils.py#L49) that further improved the point cloud quality. This is run on the object point cloud during inference: https://github.com/NVlabs/GraspGen/blob/main/grasp_gen/grasp_server.py#L187

Hope this answers your question!

adithyamurali, 2025-08-03:
Hi @Guptajakala I'll be closing this, feel free to reopen if there are other issues!

closed 2025-08-03 by adithyamurali