NVlabs/GraspGen #9: Antipodal Grasp data generation in isaaclab

aditya-nisal, 2025-08-01:
Hello,

Thank you for your contribution to this project and open sourcing it. When are you planning to release the code for data generation using pinch grippers in isaaclab? 
Another question I had was that, your paper mentions that grasp gen works better than contact graspnet in cluttered scenes. However, contact graspnet is trained on cluttered scenes but grasp gen is not. If we do not have a vision based planner, in that case, I wonder why should one choose grasp gen over contact graspnet for cluttered scenes? If you could answer these questions of mine would be really helpful.

Best 
Aditya

aditya-nisal, 2025-08-08:
Hi @adithyamurali , thank you for responding my questions. Waiting for the new repo with the data generation part.
My bad, I was thinking something else when I mentioned the vision based planner, no need to worry about it. 
All my questions are answered and I am closing the issue. Thank you for your work!

closed 2025-08-08 by aditya-nisal

adithyamurali, 2025-10-03:
Hi @aditya-nisal 
FYI The grasp data generation repository has been released separately: https://github.com/NVlabs/GraspDataGen
The datasets generated in that repo is interoperable with the training examples here.