NVlabs/GraspGen #10: Robotiq 2f-140 URDF

martinmatak, 2025-08-06:
Hi,

Can you please provide the URDF for the Robotiq 2f-140 gripper with the frame for which the grasps are generated? 

Thanks!

martinmatak, 2025-08-07:
The USD of the gripper that was used in the simulator would also be useful

martinmatak, 2025-08-08:
I also opened a thread on this here: 

https://forums.developer.nvidia.com/t/asset-improvement-request-robotiq-2f-140/341636

adithyamurali, 2025-08-12:
Also @martinmatak just a heads up, we are also planning to release a separate Issac Lab-based repo for grasp data generation. This should have the USD assets you are mentioning above tested in Isaac Lab/Sim.

This will be for several grippers, including Robotiq-2f-140. However, this timeframe for release is 1-2 months...

adithyamurali, 2025-10-03:
Hi @martinmatak hope you are doing well! Sorry for the delay in response.
The Isaac-Lab based grasp data generation repository has been released: https://github.com/NVlabs/GraspDataGen
USD files: https://github.com/NVlabs/GraspDataGen/tree/main/bots

adithyamurali, 2025-10-04:
Closing this, feel free to reopen!

closed 2025-10-04 by adithyamurali

martinmatak, 2025-10-30:
Yes, this is exactly the URDF I was looking for, perfect! We can try out the grasps from the dataset in the real world now :) 

Is there any chance you have access to the .usd of the Robotiq 2F 140 gripper you used during data collection?