NVlabs/GraspGen #15: Unable to download checkpoints file from HF

harumo11, 2025-08-24:
Hello, Developers. 

I tried to download checkpoints using [this command](https://github.com/NVlabs/GraspGen#download-checkpoints) in README.md.
However, I got following error.

```
11:39:55 ((.venv) ) harumo@harumo-desktop GraspGen ±|main ✗|→ git clone git@hf.co:adithyamurali/GraspGenModels
Cloning into 'GraspGenModels'...
git@hf.co: Permission denied (publickey).
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
```
Could you check the access rights?

This is not fatal issue, because I can download checkpoints from HF directory using following commands.

[image]

adithyamurali, 2025-08-29:
Thanks @harumo11  for the feedback! I'll update this link in the readme tomorrow

adithyamurali, 2025-08-30:
thanks once again for catching this. This has been updated 820d8329127f8afdd1785390784db517a3989c00

closed 2025-08-30 by adithyamurali