NVlabs/GraspGen #24: Is there a simple way to try this out in Isaac Sim in simulation?

rabhishek100, 2025-09-17:
Is there a simple way to try this out in Isaac Sim in simulation?

adithyamurali, 2025-09-22:
Hi @rabhishek100 currently we do not have an example for this.

We will be releasing our IsaacLab/IsaacSim based grasp data generation package soon. In there, you could technically run a inference through the model, and evaluate them in IsaacLab. Will keep you posted

adithyamurali, 2025-10-03:
Hi @rabhishek100 the IsaacLab based grasp data generation package has been released: https://github.com/NVlabs/GraspGen

While this is technically not what you asked for :) you could consider running inference with the model, saving the grasps to the YAML format described in that repo, and running the `grasp_sim.py` script for [Grasp Validation](https://github.com/NVlabs/GraspDataGen/blob/main/docs/components/grasp-sim.md).

Please re-open this issue if needed.

closed 2025-10-03 by adithyamurali